Printrbot/printrboardmodernmarlin #1: flashed successfully but motors don't move

martinich, 2018-11-19:
BACKGROUND:  I recently inherited a PrintrBot Simple Metal from a hack bin (no heated bed, Rev F5 2015).  Initially it would connect randomly (like 0.01% of the time in cura) and then I would be able to jog the motors and set the hotend feedback loop, which told the board is likely okay but there is an issue with the software.  Also, microSD with a auto0.g file never worked. 

PROBLEM:   I was able to successfully flash the board using several versions of your firmware .hex files using ATMEL FLIP (using a Win 10 LattePanda SBC computer).  The good news is it now connects via USB Serial COM8 very quickly and 100% of the time which was never happening before.  The bad news is although it connects, I can not jog the motors or set the hotend temperature.

Do you have any suggestions as to where the issue might be?  I am quite new to both AVR microcontrollers and 3d printing.  So far I was able to make some good progress following your instructions but maybe I am missing something obvious regarding setting up the stepper motor currents or something (before compilation).  I have not compiled anything from source code and I'm wondering if I did what I might change.  

I do have a little serial analyzer and pocket oscilloscope available if I need to debug the board itself, but my hope is that the board is OK and this is a software issue.  

Thanks for your help or suggestions!

drphil3d, 2018-11-20:
That’s interesting your the second person with that issue

drphil3d, 2018-11-20:
Open the terminal and type
M502
M500

This will clear and update the EEPROM
Let me know if that works

drphil3d, 2018-11-23:
if you flash firmware with heated bed and don't have one the printer may hault as safety measure. 
it also may be possible that you flashed your board with the wrong Rev firmware which has different pin assignments. I feel like I've run into this issue before. type M119 to check endstop status, if you have a newer board the stepper current is adjusted with M909 and M910 to set, I documented this in the readme.

closed 2018-12-14 by drphil3d