Printrbot/printrboardmodernmarlin #2: Can we work a flavor out for the Smalls?

asalottin, 2018-11-21:
Hey Phil, hope this finds you well.

Not really familiar with Github so please excuse me if this should not have been reported as an Issue.

I've attempted a few different firmwares for my Smalls, but none worked -- details below. Can we work something out for the Smalls? Mine has the extended/heated bed and I'd like to add an LCD to it.

Here's what I've tried with no success so far:

Attempt 1 --> RevF-v0.5 PlayV2-HB-Y_DIR_NORMAL_170x220_LCD.hex. This has the same bed size, though reserved X/Y, and apparently everything I want, but FLIP would not load it properly, throwing out a 'out of range' error.

Attempt 2 --> RevF-v.05-Simple MetalHB-150x150-NoLCD_inverting Probe.hex. This did flash properly, the heated bed was recognized, X and Y would home properly, but Z went away from the bed and would not home.

Attempt 3 --> v.04-Simple Metal-RevF-NoLCD_Non Inverting Probe.hex. This did flash properly as well, homed fine in all directions, auto-bed-levelling with G29 was also fine. However, the heated bed was not recognized (expected) and I could not change the bed size with M211.

Looking forward to be of assistance so we can port this newer Marlin to the Smalls out there!

Kind regards,

Antonio

drphil3d, 2018-11-21:
I will compile firmware for that model with heated bed.

drphil3d, 2018-11-23:
can you tell me what the dimensions of your machine are? also what Rev printrboard do you have? I will try and get the LCD support on there, unfortunately the printrboards flash memory is only 128k and the most recent version of Marlin with linear advance plus all the features is slightly oversized. LCD support takes up a ton of room.

asalottin, 2018-11-24:
Hi Phil, absolutely. The heated bed upgrade for the Smalls gives you 200mm on the X and 150mm on the Y; also, I have 150mm on the Z which is the standard. My machine has a Printrboard Rev F5. All-in for testing it and reporting back with my findings. Thank you for your willingness to help!

asalottin, 2018-11-24:
Let me amend my last comment... In the whole spirit of help-me-help-you (and perhaps many more), I continued my testing and flashed my board with other firmwares. This particular one (https://github.com/Printrbot/Marlin/releases/tag/RevF-Version-5) gets the LCD to work; LCD being a 20x4 Reprap Discount screen. This is an older Marlin release and does not home properly on the Z (probe needs inverting), so I cannot use it. It does let me change the bed size though. To get the screen to work, I wired it as described on this video https://youtu.be/IkmpZxqOHW4. For ease of reference, I'm enclosing the mapping of the connections from the Printrboard to the LCD. This firmware does not enable the on-screen SD card, but it does let me use the embedded SD card slot on the Printrboard, which is good enough. Hope this helps! Cheers, Antonio

Printrboard (Rev. F5) | to | LCD (20x4 Reprap Discount)
Pin 1 (GND) | ---> | EXP1 - Pin 9
Pin 2 (5V) | ---> | EXP1 - Pin 10
Pin 3 | ---> | *Not Connected*
Pin 4 | ---> | *Not Connected*
Pin 5 | ---> | EXP1 - Pin 8
Pin 6 | ---> | EXP1 - Pin 7
Pin 7 | ---> | EXP1 - Pin 6
Pin 8 | ---> | EXP1 - Pin 5
Pin 9 | ---> | *Not Connected*
Pin 10 | ---> | EXP1 - Pin 3
Pin 11 | ---> | EXP1 - Pin 4
Pin 12 | ---> | EXP2 - Pin 5
Pin 13 | ---> | EXP2 - Pin 3
Pin 14 | ---> | EXP1 - Pin 2

drphil3d, 2018-11-25:
I just posted new firmware for the smalls, I did not include lcd support on this, I'm still working on optimizing space for it.

> Hi Phil, absolutely. The heated bed upgrade for the Smalls gives you 200mm on the X and 150mm on the Y; also, I have 150mm on the Z which is the standard. My machine has a Printrboard Rev F5. All-in for testing it and reporting back with my findings. Thank you for your willingness to help!

asalottin, 2018-11-25:
Sounds good Phil. I'll give it a try and report back to you! Cheers, Antonio

closed 2018-12-14 by drphil3d