Printrbot/printrboardmodernmarlin #5: Printer bed size soft endstops

pfremm, 2019-01-12:
I am trying to change the bed size and before M211 Z250 would work fine.  That seems to not work anymore.  What am I doing wrong?

JSolendorf, 2019-01-13:
Looks like you have to set the Software Endstops in the firmware itself.  I'm glad you mentioned this. I discovered my Z height was set to 200 and I needed it to go to 230.  I had to modify the firmware to set my bed size and Z height.  You could disable the software endstops with M211 but you lose the safety it provides.

>> doc for this is here...
**http://marlinfw.org/docs/hardware/endstops.html**
Software Endstops
Typically 3D printers are only equipped with hardware endstops on one side of each axis (Minimum or Maximum of the respective axis). As discussed above this is used to determine the starting point (origin) of the machine coordinate system.

In order to also protect the other side of the axes software endstops should be defined in the firmware via the #define MAX_SOFTWARE_ENDSTOPS / #define MIN_SOFTWARE_ENDSTOPS directive. This then uses the value from #define [XYZ]_MAX_POS / #define [XYZ]_MIN_POS to determine the maximum distance between the physical endstop and the software commanded stop of the axis. Software endstops can be (de-)activated via the M211 G-code.

drphil3d, 2019-01-14:
I don't like having max software endstop disabled, all it takes is the wrong gcode to damage your printer

pfremm, 2019-01-20:
It was nice before where this could be set via gcode.  Was this lost in the firmware upgrade some how?

drphil3d, 2019-01-20:
Offsets don’t work on newer marlin firmware.

pfremm, 2019-01-20:
Can someone create a simple metal firmware with the super x, and super z them?

drphil3d, 2019-01-20:
Sure I can put that together. The smalls uses the same exact firmware fyi

closed 2019-02-01 by drphil3d