Printrbot/printrboardmodernmarlin #6: Which file to use for Metal Plus (Rev F4 board)?

MemeticParadigm, 2019-01-13:
Trying to figure out exactly what approach to take for updating my dual extruder Metal Plus. Do I just use the most recent RevF file (RevF-0.6.5_PlayHB_200x100_Inverted_Probe.hex) and then load the previous eprom settings, or what?

drphil3d, 2019-01-14:
I don't have the metal plus dual extruder firmware on here yet, I can work on that for you in the coming weeks.

MemeticParadigm, 2019-01-14:
That would be enormously appreciated =)

Zakattacks, 2019-01-28:
I uploaded the stock Plus firmware but It is as of yet untested. I'm not familiar with the Plus dual extruder, but you're welcome to download it and configure it.

rdragonrydr, 2019-08-02:
I have the same question --admittedly I'm not certain of my board revision, and need to check that first-- but I can't seem to find firmware for the Metal Plus Dual specifically?
Does it matter if I have the green inductive probe rather than the metal tubular one?

ollien, 2020-04-06:
@drphil3d Is there support for the Metal Plus Dual Extruder at this point? I have a RevD board, and though I can try to configure it myself, I'd much rather use something official if I can.