Printrbot/printrboardmodernmarlin #7: How to set min/max position on Simple Metal Extended?

ColinTravis, 2019-01-14:
Currently, my M211 reports my Min and Max at

echo: Min: X0.00 Y0.00 Z0.00

Max: X125.00 Y203.00 131.00

When the documentation states it should be at 
X152.40 Y152.40 Z152.40

The extended bed should bump it up to about X250, as well.

However, sending an M210 results in "Unknown Command", and sending M211 just reports the value of the soft endstops. 

What are my options?

ColinTravis, 2019-01-14:
I'm just seeing the other issue reporting about software endstops. Closing this.

drphil3d, 2019-01-14:
Unfortunately with marlin 1.1.9 I can't find a way to easily redefine the machine dimensions easily like you could in older versions of marlin. The smalls firmware will work the same on the simple metal

ColinTravis, 2019-01-14:
Hmm, it looks to be a better option, but the dimensions are off for the simple metal extended. The bed changes the X to be 250. I tried compiling the firmware with an updated Configuration.h, but I keep getting an error thrown stating:

`Marlin/pins_PRINTRBOARD.h:65:4: error: #error "Oops! Make sure you have 'Teensy++ 2.0' or 'Printrboard' selected from the Tools -> Boards menu` 

I'm using the marlin-config gui to compile.

Zakattacks, 2019-01-28:
According to https://github.com/Printrbot/printrboardmodernmarlin/issues/8 you have to manually change the teensy20 to teensy2. However, I have not had this issue.

drphil3d, 2019-12-26:
https://github.com/Printrbot/printrboardmodernmarlin/releases/tag/2.0.1

closed 2019-12-26 by drphil3d