Printrbot/printrboardmodernmarlin #8: No Heated Bed or Thermistor for Simple Metal?

ColinTravis, 2019-01-23:
I'm currently using the Smalls firmware on my Simple Metal, and neither Cura or Prontrface can detect my heated bed.

When I load up Prontrface, anytime I fire up the heated bed, I just get invalid command errors.
[image]

drphil3d, 2019-01-23:
I may not have put heated bed support on that model. I will get some new
firmware uploaded I’ve been putting it off but people keep asking.

ColinTravis, 2019-01-24:
Awesome, this sounds great! Is there any way we can compile the 1.1.9 firmware ourselves? I know I've been having some issues with it compiling that I didn't have with 1.1.8.

Zakattacks, 2019-01-27:
The easiest way to compile the firmware is to use <a href="https://github.com/akaJes/marlin-config/releases/tag/v2.10.7">marlin-config</a> to compile the firmware. You have to download the latest release of <a href="https://github.com/MarlinFirmware/Marlin/archive/1.1.x.zip">Marlin</a> and then copy and paste the Configuration.h and Configuration_Advanced.h files into the Marlin directory. These can be found in the Source directory under each printer. Select your Marlin directory in marlin-conf and make you modifications. To compile, select teensy20 under the compile menu, then click compile. <b>The built in flasher will not work</b>.  You have to seek out the .hex file. It's called firmware.hex and is stored in the .pioenvs directory within your Marlin directory. You should be able to flash this .hex per the <a href="https://reprap.org/wiki/Printrboard#Loading_Firmware_.28Windows.29">RepRap Wiki</a>.

Zakattacks, 2019-01-27:
I'm almost done reorganizing the repo, it should be done sometime today.

ColinTravis, 2019-01-28:
That did the trick! Compiled like a champ. Had to rename the teensy board as stated above, but it worked without fail. Haven't flashed it yet onto my printer, but this was insanely helpful!

kolakidd, 2019-01-28:
One thing I found, if it's a simple metal then you must use the config file from printrboardmodernmarlin/Simple_Metal/Simple Metal/ (It seems to be specifically for Rev F)
NOT the printrboardmodernmarlin/Simple_Metal/Simple Metal 2018-11-12 13-19-56/ version

I managed to flash it to the board this evening. I disabled LCD features in order to save space.

My only issue, and it's a biggie is that the inductive probe is behaving weirdly. I have an aluminium bed which it is not detecting, but it is detecting my metal pliers. by detecting, I mean the led is coming on, but it doesn't seem to be triggering the stop. Other endstops are triggering perfectly.

Zakattacks, 2019-01-28:
Open a new issue if you need more help.

closed 2019-01-28 by Zakattacks

wBrhy2, 2019-04-01:
> Thank you! I went through the process of compiling the hex above. I haven't had a chance to install it, but the process works perfectly. Only one issue I detected which may help others.
> 
> In the platformio.ini file (Selected Marlin-Config folder>Marlin>
> The Teensy board ID is incorrectly labelled which results in an Unknown board ID message when compiling. In order to correct this you need to change
> board = teensy20pp TO board = teensy2pp

where do you change this name in the malrin configurator?

kolakidd, 2019-04-01:
Hi, on Mac the file path to the In the platformio.ini file is <Selected Marlin-Config folder>/Marlin/platformio.ini

wBrhy2, 2019-04-01:
> One thing I found, if it's a simple metal then you must use the config file from printrboardmodernmarlin/Simple_Metal/Simple Metal/ (It seems to be specifically for Rev F)
> NOT the [...]

> Hi, on Mac the file path to the In the platformio.ini file is /Marlin/platformio.ini
> 
> I disabled the LCD through the GUI and it seemed to do the trick, the GUI is editing the configuration.h file.

thanks! it was easy to find, same place on windows.

For the LCD, do you know which option it was? There are just so many rows, I don't know if I've checked the right one or not despite trying to carefully go through every row twice now. Its hard to focus that long.

wBrhy2, 2019-04-01:
> The easiest way to compile the firmware is to use [marlin-config](https://github.com/akaJes/marlin-config/releases/tag/v2.10.7). You have to download the latest release of [Marlin](https://github.com/MarlinFirmware/Marlin/archive/1.1.x.zip) and then copy and paste the Configuration.h and Configuration_Advanced.h files found in the Simple>Source>XXXX directories into the Marlin directory. These can be found in the Source directory under each printer. Select your Marlin directory in marlin-conf and make you modifications. To compile, select teensy20 under the compile menu, then click compile. The built in flasher will not work. You have to seek out the .hex file. It's called firmware.hex and is stored in the .pioenvs directory within your Marlin directory. You should be able to flash this .hex per the [RepRap Wiki](https://reprap.org/wiki/Printrboard#Loading_Firmware_.28Windows.29) instructions.

why do we want teensy20 for the printrbot simple metal instead of at90USB1286_CDC