Printrbot/printrboardmodernmarlin #9: heated bed tuning for thermal runaway.

pfremm, 2019-01-29:
@Zakattacks I think I as talking to you on reddit about heated bed tuning.  I definitely have some ripple when I try to use my heated bed.  Thought if maybe we discuss on here it might be helpful to others or add to readme.

Zakattacks, 2019-01-29:
Copied and pasted from our reddit conversation:

>  If you're getting a "Thermal Runaway Protection" or similar error it could be many things. I had an issue that the Bangbang/PID settings for the bed were so wrong that it triggered the thermal runaway protection. The diagnosis for this is a strong ripple on the heated bed like this: https://i.imgur.com/29dH6AM.png Then you can enable bed PID if it's not already enabled. You then have to tune the bed to get that line flat.

I have seen this issue in many of my Printrbots, and have enabled PID temp bed on all of the firmware I've uploaded here. So in theory all the firmware above V0.7 in this repo should have PID temp bed enabled and you should be able to tune your bed by running <a href="http://marlinfw.org/docs/gcode/M303.html">M303</a>. I will typically use <code>M303 E-1 S70 C8</code> but everyone's use case is different. If this throws an error code you need to enable  PIDTEMPBED and recompile.

Zakattacks, 2019-02-01:
Were you able to try this?

pfremm, 2019-02-01:
I have a version from the v.3 branch installed so it would install on my ref F board.  That seemed to have bang bang enabled.  The build in releases seemed to be to large for me to load.

kolakidd, 2019-02-03:
> I have a version from the v.3 branch installed so it would install on my ref F board. That seemed to have bang bang enabled. The build in releases seemed to be to large for me to load.

Try disabling LCD stuff if you don't need it.

pfremm, 2019-02-08:
DO we have a good guide on compiling firmware?  This is beyond my current expertise.

ColinTravis, 2019-02-09:
I have a strange issue when I enable PIDTEMPBED. If it's enabled, my stepper motors don't move, if it's enabled, they work as expected.

drphil3d, 2020-01-08:
This is fixed in 2.0.1

closed 2020-01-08 by drphil3d