Printrbot/printrboardmodernmarlin #10: Firmware creation process

kolakidd, 2019-01-30:
Hey, okay, I'm going to come clean, I am a total Newb to Github. I have drafted a complete set of instructions to create the firmware for the Printrbot based off the config files here. I am happy to put it up, but I feel like I need to ask the question:

Is this okay with DrPhil, and is this the right place to put it, should it go on the printrbottalk forum or somewhere else. 

Apologies again, I've never really used Github

drphil3d, 2019-02-01:
I would happily welcome any documentation, I can create a wiki page. The documentation would be more useful here and on the reprap wiki though it’s kind of difficult to edit. Printrbot talk forum is harder to find info for most people.

drphil3d, 2019-02-01:
Here you go, you should be able to edit any of these pages, after submission I approve it and it goes up.
https://github.com/Printrbot/printrboardmodernmarlin/wiki/Compiling-Firmware

kolakidd, 2019-02-03:
Thanks! I have uploaded a guide, comments, corrections, fire and hatred all welcome.

JSolendorf, 2019-02-04:
1.  Thanks for doing this.
2.  Not sure what Very NB means>>     **(Very NB for Simple: select the config file for REVF!)**
3. There is another page in the WIKI that can be removed: 
  https://github.com/Printrbot/printrboardmodernmarlin/wiki/Compiling-Firmware

4.  Don't forget to make this a link;    **Download and install PlatformIO Download PlatformIO here**

kolakidd, 2019-02-05:
No problem, changes have been made. 
nb - changed to very important, it was showing my age and schooling, it's an abbreviation of latin from notary bene (note well), it's an old habit.

closed 2019-02-07 by Zakattacks