Printrbot/printrboardmodernmarlin #11: Compiling Error

Jozer99, 2019-02-01:
I am having a problem compiling and flashing the firmware per your directions in Windows.  I installed Marlin-conf using the executable on their Git site, and successfully import one of your configurations into Marlin-conf.  I set it to compile using the "at90USB1286-DFU" setting, then click compile.  

The first time I do this, the PlatformIO terminal pops up and shows it downloading a bunch of things (the last successful package being "LibraryManager: Installing c1921b4"), until it stops with the error:
`Error: Unknown board ID 'at90USB1286'`.  

Subsequent attempts to compile just give the error:

`Processing at90USB1286_DFU (platform: teensy; board: at90USB1286; framework: arduino)
Error: Unknown board ID 'at90USB1286'
Done.`

I've tried this with Marlin-conf versions 2.10.7 and 2.10.5, with the same result.  Any tips?

drphil3d, 2019-02-01:
you have to select tiny20 from the menu, make sure you selected marlin 1.1.9 as far as I'm aware 2.0 wont compile

Jozer99, 2019-02-01:
Changed to "teensy20", I get a very similar error:

`Processing teensy20 (platform: teensy; board: teensy20pp; framework: arduino)

Error: Unknown board ID 'teensy20pp'

Done.`

kolakidd, 2019-02-01:
Solution to teensy20 issue is here: https://github.com/Printrbot/printrboardmodernmarlin/issues/8

Jozer99, 2019-02-01:
I've also tried running Marlin-config in Ubuntu 18.04 (previously used Win10).  I get the same error.  Can you provide directions for working compilation?  If not, can I provide you with a Configuration.h file to compile?

Jozer99, 2019-02-01:
@kolakidd That seems to have resolved the unknown board id error.  Now I get a bunch of warnings and eventually an error. I am attaching the log to this post.  I haven't changed anything but the bed size and enabled SD support.  
[errors.txt](https://github.com/Printrbot/printrboardmodernmarlin/files/2822872/errors.txt)

wBrhy2, 2019-04-02:
> Solution to teensy20 issue is here: #8

why do we want teensy20 for the printrbot simple metal instead of at90USB1286_CDC ? when flashing with FLIP, I use at90USB1286

drphil3d, 2019-07-01:
the options you choose in flip vs compiling using platform IO are different but same processor type

closed 2019-07-01 by drphil3d