Printrbot/printrboardmodernmarlin #13: Heated bed not enabled in New .8 revision for simple metal

fintechcommunications, 2019-03-08:
I tried flashing the new .8 revision and noticed heated bed is not enabled in marlin. I love the work that you do. Made my old printrbot awesome.

drphil3d, 2019-03-08:
just flash your printer with the RevF-0.8_Printrbot_Simple-Metal_HB_152.hex firmware

fintechcommunications, 2019-03-08:
That was the version I tried. Heated bed not showing in Pronterface or cura 15. I'm not getting a thermistor reading. Firmware below is working perfectly fine. I'm running a E3D hotend with the stock printrbot thermistor. I would like to upgrade my hot end to E3d cartridge style. I currently have tried 
RevF-0.8_Printrbot_Simple-Metal_E3D_HB_152.hex  (No heated bed reading)
RevF-0.8_Printrbot_Simple-Metal_E3D_152.hex (No heated bed reading)

This version works
RevF-v.05-Simple MetalHB-150x150-NoLCD_inverting Probe.hex

fintechcommunications, 2019-03-08:
Your firmware is awesome by the way. My printrbot prints better then any of my other machines.

drphil3d, 2019-03-09:
Yeah... I don't think most people understand just how important firmware is to print quality. I will compile it right now.

drphil3d, 2019-03-09:
Here you go. Make sure to run PID autotune for the hotend and heated bed if your using E3D

drphil3d, 2019-03-09:
https://github.com/Printrbot/printrboardmodernmarlin/releases/tag/v1.0

drphil3d, 2019-03-09:
You should consider upgrading to Cura 3.6 it will improve your print quality

fintechcommunications, 2019-03-09:
Awesome, thank you very much. Not sure what your doing with this firmware that makes it print so good.

fintechcommunications, 2019-03-09:
I wish they would have kept Pronterface part of Cura. I wish they had better com support in the new versions. I have octopi but I prefer to print directly from my usb port. What do you like better Cura or Simplify3D?

drphil3d, 2019-03-09:
I use the Octoprint plugin available in Cura 3.6 which sends the print directly from cura. With modern firmware the printer is reliable enough that there's really nothing to mess with. If I really need access to the web ui then I can just click the button.

I don't really like simplify3D at this point.
You have to understand that machine control and slicing are two separate things. In the old days it was normal for a printer to be driven by a desktop computer but now most all printers are wireless.

IceSL slicer is pretty awesome https://icesl.loria.fr/
Pathio is up and coming https://pathio.xyz/ 
[image]

closed 2019-03-09 by drphil3d