Printrbot/printrboardmodernmarlin #18: Unable to change Z probe offset with M851

evenlund, 2019-04-12:
M851 is not working like M212 used to work on original printrbot firmware for my Printrbot metal simple. The Z location is not changing at all when homing with G28 or in octopi.

I flashed the hex-file with E3D hotend and heated bed. Next I used M502 and M500 to restore settings since others that have had issues with M851 not working had forgotten this. I used M500 after setting the new probe Z offset position and the new position is correctly shown with M501. When trying to set offset postion lower than -2 i get this error message in octopi: "Error reported by printer. Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: ?Z out of range (-2 to 20)". This is probably due to settings in firmware and should be fine if probe setting is working.

Is this a bug in the firmware that or can anyone confirm that they have used M851 successfully to set Z probe offset? Is there a solution?

drphil3d, 2019-04-12:
What's the exact command that you used?
http://marlinfw.org/docs/gcode/M851.html

I recommend to not even mess with setting the offset manually and instead adjusting it while the first layer is printing using M290

http://marlinfw.org/docs/gcode/M290.html

evenlund, 2019-04-14:
I used home z in octoprint (M28 Z) and measured the height of the nozzle with a paper, similar to what Brook and Tom did in their tutorials. I then did M851 Z-0.2 similarily to what you would do previously with M212. I then repeated homing Z, measuring height of the nozzle and decreasing probe offset. With these steps I could not see any diffence in the distance between nozzle and bed.

I see from the Marlin M851 documentation that there are some additional steps to what I did previously (ie. M401 deploy probe), and I followed these after adjusting the probe height according to original Printrbot instructions. I am unsure whether this fixed the problems I had, but I am now able to get the first layer correct by final adjustment with M290. Do you know whether M290 height is stored between shutdowns of the printer?

wBrhy2, 2019-04-16:
https://github.com/Printrbot/printrboardmodernmarlin/issues/17

closed 2019-07-01 by drphil3d