Printrbot/printrboardmodernmarlin #19: lcd pinout

wBrhy2, 2019-04-19:
Can't seem to get the LCD to work, but I'm not sure if its a faulty LCD or if I screwed up so... two questions

1- is the 1.0 firmware provided LCD enabled? (I am under the impression it is)
2- what is the pinout mapping you used?

Thank you!

Zakattacks, 2019-04-19:
I know that it takes some work to integrate LCD support as there is barely enough space. However this isn't my area of expertise so I'll let @drphil3d (who created the hex) explain

drphil3d, 2019-07-01:
There isn't support for LCD because the latest version of marlin requires 98% of the very limited 128k of memory the Printrboard has. So the trade of is either an easy to use printer that gives high quality prints or LCD support and few features.

closed 2019-07-01 by drphil3d