Printrbot/printrboardmodernmarlin #21: Simple metal heated bed not reading out

wBrhy2, 2019-05-23:
Hello, I use octoprint and have been playing with different firmware configurations for the past few weeks. For some reason I can't get the heated bed temperature to be read (and as such no heating is possible). I have tried setting it myself and using the firmware v1.0 provided here.

I have tried measuring the thermistor, and it comes out at 100k ohm as it should.

Any ideas on how to troubleshoot this some more? Is there some way I can use the gcode terminal to probe stuff?

Thanks

wBrhy2, 2019-06-07:
Sorry to bump my own message, but can anyone confirm a known firmware that works with the heated bed so I can test if it's my board?

kolakidd, 2019-06-08:
I have a metal simple with a heated and extended bed, also running off Octoprint. The firmware and method I used is in the wiki, under compiling firmware.

It does sound like a firmware issue, but very stupid question: have you tried to read the temperatures by direct usb connection to pc with Octoprint?

wBrhy2, 2019-06-08:
So you are using firmware? That's what I'm using right now.
RevFv1.0_Printrbot_Simple_Metal_HB.hex

What do you cm by direct usb connecting? My printer is connected via USB to
a rpi running octopi.

wBrhy2, 2019-06-08:
For the purpose of testing I connected via USB to my android device using gcodeprintr, it did not get any readings for the heated bed, but also returned a command not recognized error when I tried changing the temp... I guess I'll try reflashing again.

wBrhy2, 2019-06-08:
Alright, I reflashed and it's fine..... I don't get it since I tried like 6 times, but after re-downloading it fresh then it worked. Thanks!

drphil3d, 2019-07-01:
sorry I didn't see this issue until now for some reason didn't get a notification.

closed 2019-07-01 by drphil3d