Printrbot/printrboardmodernmarlin #23: New FW an a Simple Metall: failed at 0x00000

3kmch, 2019-07-20:
A couple of day ago I tried to flash the "RevFv1.0_Printrbot_Simple_Metal_.hex" to my Simple. But Flip refused to do it. "Jozer503" describes the same behaviour:

> I tried to install Modern Marlin on it using the wiki directions with FLIP. I got my printer into DFU mode and the driver installed.
> 
> When I tried to run an Erase, Zero Check, Write, Verify cycle, the Erase step "completed successfully", but the zero check failed at 0x00000. When I reboot the board, I get back into the original firmware. If I skip the Zero check, it also "writes successfully", but fails verification, and again, I boot back into my original firmware.

http://www.printrbottalk.com/forum/viewtopic.php?f=22&t=11895

Any idea what went wrong?

wBrhy2, 2019-07-20:
Did you move the jumper?

3kmch, 2019-07-20:
Yes, I set the jumper to "flash mode".

drphil3d, 2019-08-20:
Are you right clicking saving the hex file directly from github or did you download as a zip? If you right clicked and downloaded you will end up with an html document inside the file your trying to use. Check the file size should be around 120k 
[image]

Try these settings as well toggle between EEPROM button and try flashing multiple times. 
[image]

What Rev Board are you using? 
When you press the reset button with the header jumped does the printer show up as a DFU device? make sure your using the FLIP 3.4.7 Programs folder as the source for the drivers.

3kmch, 2019-08-31:
Thank you for your support. It worked this time. :)

Unfortunately the LCD did not work any longer. Do I habe to rewire it? And if I get it back to work, is there a menu where I can adjust the z offset?

drphil3d, 2019-08-31:
The reason there's no lcd support is because Marlin 1.1.9 with all the new features which greatly improves print quality consumes all the available program memory. The lcd is a less common so I had to choose a compermise between print quality and ease of use vs lcd. The only way to make it fit is if you use an ICSP programmer to overwrite the bootloader sense that's what's consuming a good chunk of memory. I know with the older printrboard they used a different type of bootloader which takes less memory.

Actually there is one other option. I don't remember the command offhand but it tells avrdude to force the program to load disregarding the warning about space.

3kmch, 2019-09-01:
OK, in this case I skip modern marlin and stay with setting the z offset by M212.
Thank you anyway :)

closed 2019-09-01 by 3kmch