Printrbot/printrboardmodernmarlin #24: Simple Metal Post auto-leveling rapid move to X0, Y0, and Z up 10mm too rapid

unixunion, 2019-07-21:
Hi, 

Can you possibly remove the POST auto-leveling rapid movement to 0,0,10 as its too fast for my machine, and its not nice having such a movement hard coded in post autoleveling. I cant seem to get the firmware compiled myself, as platformio is giving me hell to try and compile for at90USB1286_DFU. 

Thanks.

drphil3d, 2019-08-20:
Sure, I can do that. You have to use Teensy20 to compile firmware. Personally I use Marlin-Config to compile mainly for the reason that all the dependencies for the development environment are already setup. https://github.com/akaJes/marlin-config/releases

Here is more information about compiling firmware
https://github.com/Printrbot/printrboardmodernmarlin/wiki

drphil3d, 2020-01-05:
I removed the script from the latest version 2.01 available under releases.
If you want to compile it yourself the files are available along with the platformIO environment variables to make it compile. Choose option AT90usb1286_DFU

closed 2020-01-05 by drphil3d