Printrbot/printrboardmodernmarlin #26: Simple Metal HB - Board RevF5 - Ymin Crash?

Gizmo1403, 2019-08-20:
Hi All!

I recently downloaded RevFv1.0_Printrbot_Simple_Metal_HB.hex for my Simple Metal with heated bed and board RevF5.  Everything is original to the printer (I have made no modifications).  I think this is the right firmware to use?  I did manage to get the firmware flashed to my printer; however, I am running into a minimum Y issue.  I am not a programmer, but what I think is happening, is the minimum Y value is not 0 within this or any other .hex file for the Simple Metal.  The minimum Y is -2, thus when trying to setup the auto-level functions, specially when G29 is used, the minimum Y is trying to go beyond zero (-2mm according to the firmware).  This causes a very unpleasant noise, as the Y axis is trying to go beyond the print platform.  Is the above correct?  Or am I doing something wrong?  Is there a way to fix this?

drphil3d, 2019-08-20:
Is your Y axis moving in the same direction to home as the previous firmware? I noticed some printers have different stepper motors and they are wired differently so they go the opposite direction.

Gizmo1403, 2019-08-21:
Yes, but it depends.  I am using Cura 15.04.6.  Home for the previous
firmware for all XYZ axes used to be the back left position (X=0, Y=152.4,
Z = Calibrated point).

With the new firmware if I just home the Y axis, it homes to Y=152.  If I
just home the X axis it homes to X=0.  If I home everything (or home just
the Z axis) the printer will go to the middle (X=56, Y=76, Z=0).  Keep in
mind I have not calibrated the automatic bed leveling.  Also I do not have
the printer setup inside Cura to have the home position "Machine Center
0,0".

The issue of the Y axis crashing comes up when setting up the automatic bed
leveling, specially sending the G28 command first then the G29 command.  At
the end of the G29 command the Z axis rises from the middle of the back
edge, then it appears it's trying to move to the X=0, Y=0 position but it
seems to slam into the Y=0 (or maybe it's trying to go to Y = -2, which
would be too far?).

Also, in general, when I type the M211 command, this is what I
see:  echo:Soft endstops: On   Min: X0.00 Y-2.00 Z0.00  Max: X152.00
Y152.00 Z152.00

Hope you guys can help.

drphil3d, 2020-01-05:
I fixed this in the latest release available under releases tab. Marlin 2.0.1

closed 2020-01-05 by drphil3d