Printrbot/printrboardmodernmarlin #27: Z probe Logic reversed after firmware upgrade

Dragon2040, 2019-09-17:
Upgraded the firmware on my Simple Metal RevD. Since doing so the logic on the probe is now reversed. 
M119 shows Triggered when no metal is detected, Open when metal is detected. This of course means when homing Z, it goes up then stops.
Rolled back to an earlier firmware, issue goes away and able to print normally.

Are you able to compile an updated file with the line change for the z probe?

Thanks.

drphil3d, 2019-12-26:
I'll have to put that on my todo list

closed 2019-12-26 by drphil3d

drphil3d, 2019-12-26:
I compiled it for you, please test and let me know how it works
Marlin_2.0.1-Custom_Printrbot_Simple.Metal_RevD.Z-Probe.Inverted.Logic.hex
https://github.com/Printrbot/printrboardmodernmarlin/releases/tag/2.0.1

GrumpyMetalGuy, 2020-05-27:
Hi, I just tried this firmware, and the Printrbot refused to connect to the Raspberry Pi running OctoPrint. Flashing back to the previous firmware works, but still have the inverted Z probe problem. Is it possible to try rebuilding the firmware?