Printrbot/printrboardmodernmarlin #28: Heated Bed and Z-height notes

kolakidd, 2019-09-24:
I recently recompiled the latest Marlin for my Printrbot. Just for the recored over time I've changed the hot end to an E3D V6, extended the bed and added a silicone heat pad from Aliex Express. 

Two observations that may be helpful to others. 

Heatbed temperature and failures:
The heatbed kept on failing and killing the firmware operation. I played with extending WATCH_TEMP_PERIOD to 120 in Configuration_Adv. This helped, but the issue kept on occurring. I then noticed that MAX_BED_TEMP was set to 150, I changed this to 255, which completely resolved the issue and the bed is heating up as it did with the original firmware.
- The problem here was that the bed was simply taking longer to heat on 150

Extruder Z-height
This may just be my ailing memory, but I seem to remember Z being at the correct height when the PB is homed. This is no longer the case, when I home the bot it is around 1mm from the bed. When I set the Z using M851 the PB only seems to use the new correct value when printing. So you basically set the Z height, and then start a print and see if it is the right height. I actually found an easy way to do this using Octoprint and a plug-in which enables a dry run. I put a piece of paper on the bed, and start the dry run. At around the point where I can move the paper, but still feel that the paper is being slightly gripped, I know I am around the right height.

drphil3d, 2019-10-28:
Thats intresting, I was also having this seemingly random issue when printing where it would stop due a thermal error but I couldn't figure out why. It may be that the power supply is insufficient or the voltage is too low or it could also be from the heated bed being driven by PID in my firmware instead of bang bang. I will look into it more, please post your code if you find a solution.

drphil3d, 2020-01-05:
If you’re using an aftermarket silicone heating pad with built in thermistor you have to recompile your firmware for the thermistor in the bed, otherwise it won’t read temperature accurately. If you set the max temp to 255 that makes your printer a fire hazard sense it won’t turn off before the max rated temperature. 
You need to do a pid autotune for the bed.

drphil3d, 2020-01-05:
Oh and for the z offset, use 
M290 Z-.05 
(This feature is called babystepping)
the command is incremental so the more times you enter it the closer it moves to the bed. This only works while printing. Essentially this negates the need to manually adjust the offset, move down and repeat which is super time consuming.

closed 2020-01-05 by drphil3d