Printrbot/printrboardmodernmarlin #32: LCD don't work with Marlin 2.0

jonathan-iap, 2019-12-28:
Hi, 
I have a printrbot plus v2-1 (with wooden frame and LCD 20x4). 
So i want to change the nozzle and i need to set new parameters. 
But with Marlin 2.0 the LCD screen make whatever. 
It's display characters but all is mixing like an overflow and move all time.
Did you always try your firmware with this hardware ?
So  my motherboard is an printrboard revD .

drphil3d, 2020-01-05:
It’s possible I may have not chosen the correct LCD. Are you using the genuine Printrbot LCD?

jonathan-iap, 2020-01-06:
So i think it's a problem with platformIO but i'm not totally sure. 
I say that because the LCD work when you compile with Arduino IDE. But the memory of the printrboard seem to small for marlin 2.0 when you active LCD and sd support. 
I have open an issue on [Marlin repo ](https://github.com/MarlinFirmware/Marlin/issues/16378)

drphil3d, 2020-01-08:
Yeah there just isn’t enough memory on that processor. You can’t use more than 95% of the program memory if you want it to program successfully. The other 5% is being used by the boot loader.

drphil3d, 2020-01-08:
This is what Brook says about the LCD
 “ The connections need to be checked and you absolutely have to run a ground wire between the screen pcb and the Printrboard. That usually takes care of it. But I have fixed that with a reflash too.

Bd”

closed 2020-01-08 by drphil3d