Printrbot/printrboardmodernmarlin #34: Simple Wood 1405 (Maker) request

Bointon, 2020-01-17:
Just uploaded RevD-v.04-Simple Metal-RevF-NoLCD_inverting Probe.hex onto my Printrboard Rev D in my simple 1405 markers edition. It was the closest IMHO

The motor directions are all inverted, but this is easy to resolved by flipping the connectors. However the endstop configuration between the Simple metal and the Simple 1405 wood is different.

The x min endstop works fine, however the y and z have issues.
Stage size X and Y max are fine, but z is a little off.

**The differences I can see are:**
The simple has the endstop on y min instead of y max, also running M119 shows Y_max triggered when the Y endstop isnt pressed. The result doesnt change when the Y end stop  is pressed while M119 is run so I have no idea how the pins are routed.

The simple wood 1405 has an pnp style z endstop so it is inverted wrt the simple metal. It lights up when its close to metal and off when not. When the probe is far from metal M119 shows triggered. When close to the metal and lit up M119 shows open.

Z max is listed by M211 as 131 where as I believe Z max should be 112.

**Additional request** I have added a heated bed with a generic NTC 100K, enabling this would also be most awesome. 

So what is my request. Could you please compile a .hex compatible with the printrbot wood 1405? I have tried to do this repeatedly using the platoform IO method and the teensy/arduino method and am closer to an aneurysm than either method working.

drphil3d, 2020-01-17:
Yeah I can do that. I use marlin-config for compiling as it automatically sets up the required software. The platformIO config for marlin 1.1.9 seems to be broken, use marlin 2.0.1 instead, the source files are under releases tab. Select printrboard rev D and Use the AT90USB1286_DFU option to compile using marlin 2.0.1.

drphil3d, 2020-01-17:
if you invert the stepper motor direction, change the y home direction to + and select Ymin endstop that should do it for you.

Bointon, 2020-01-19:
Ok so I used the marlin-config to modify the configuration and configuration_adv from the RevD-v.04-Simple Metal-RevF-NoLCD_inverting Probe versions.

I compled using marlin-config (marlin 2.0.1) for the at90usb1286_dfu. This was struggling originally due to a TMC library, the log would just show a  fail and list the TMC lib. I had to modify the file platformio to ignore the TMC library. 

The at90usb1286_dfu entry now reads

[env:at90usb1286_dfu]
platform 	= teensy
board 		= at90usb1286
lib_ignore 	= TMCStepper
lib_deps 	= ${common.lib_deps}
src_filter 	= ${common.default_src_filter} +<src/HAL/HAL_AVR>
monitor_speed 	= 250000

I then took firmware.hex from the build directory ./pio/build/at90usb1286_dfu and used amtel flip to upload it to the board. To do this I followed the put the jumper on the boot pin and pressed the reset. I ticked Earse, blank check, program and verify and uploaded the hex. It completed sucessfully.

Upon rebooting I am unable to connect via pronterface @ 25000. Double checked the port number and it just said connecting.... and froze the program.

No other baud rate connected however if I power cycle and  quicky try to connect i get the following echos.

Marlin 2.0.1
echo: Last Updated: 2019-12-24 | Author: (Dr.Phil3D)
echo:Compiled: Jan 19 2020
echo: Free Memory: 3960  PlannerBufferBytes: 1568
echo:SD init fail
echo:EEPROM version mismatch (EEPROM=V55 Marlin=V74)
echo:Hardcoded Default Settings Loaded
[ERROR] Can't read from printer (disconnected?) (SerialException): call to ClearCommError failed
[ERROR] Can't write to printer (disconnected?) (SerialException): WriteFile failed ([Error 22] The device does not recognize the command.)
[ERROR] Can't write to printer (disconnected?) (SerialException): WriteFile failed ([Error 22] The device does not recognize the command.)

After that I cannot connect again without power cycling. There is a short window, about 3 seconds, where I can send a command before I get the EEPROM issue.

When I reflash the RevD-v.04-Simple Metal-RevF-NoLCD_inverting hex it connects to pronterface fine.

Is there a setting I have missed in the configuration files?

Attached is the configuration files I have used. Extensions changed to upload

[Configuration.txt](https://github.com/Printrbot/printrboardmodernmarlin/files/4083425/Configuration.txt)

[Configuration_adv.txt](https://github.com/Printrbot/printrboardmodernmarlin/files/4083423/Configuration_adv.txt)

[firmware.txt](https://github.com/Printrbot/printrboardmodernmarlin/files/4083428/firmware.txt)

Bointon, 2020-01-19:
Further to this after the board returns to hardware defaults the com port vanishes from device manager.

I have tested this on two different computers.

Bointon, 2020-01-21:
So I have found the following after uploading all of the RevD board hex on the printrboard RevD onto the simple wood 1405.

The Hex Archive/RevD-v.04-Simple Metal-RevF-NoLCD_inverting Probe uploads and operates. This as in the request had some differences to the simple wood 1405, but worked for connecting in pronterface.

I tried uploading the following hex files 
/Simple_Metal/RevD-Simple_Metal.hex
/Simple_Metal/RevD-Simple_Metal_HB.hex
/Simple_Metal/RevD-Simple_Metal_HB_LCD+SD.hex
/Simple_Metal/RevD-Simple_Metal_LCD+SD.hex

All gave the same communication issues, no sd card reading and EEPROM and Marlin version issues.

This matched my experience with the Marlin-configurator. Compiling the source for 2.0.1 using the config files found in printrboardmodernmarlin/Simple_Metal/source/revD/ kept giving the same response.

I noticed that the simple metal hex were all reporting Marlin 2.0.1 while the archived version was reporting 1.1.9.

So I attempted to compile the Marlin 1.1.9 using the Marlin-Configurator. Had to build for teensy 20 and make the mod to the Platformio to update the name:
board  = teensy20pp to board = teensy2pp

As per to post by [kolakidd](https://github.com/Printrbot/printrboardmodernmarlin/issues/8#issuecomment-458054021)

I just did a test with your config from the Marlin-configurator database and uploaded the hex using flip.

This worked well and allowed me to connect with Pronterface, no sd card error and it showed all the healthy echos.  

I used M119 and the pronterface UI to check the endstop behaviour and motor directions, then updated the configuration files accordingly. Compiled and uploaded and connected to pronterface fine.

G28 ran fine
G29 ran fine
M500,M501 and M502 ran fine

I have no idea why the 2.0.1 builds werent working for my board, but all i needed was the multipoint bed leveling to compensate for my simple wood 1405's warped gantry so 1.1.9 is perfect.

Thanks for the help and the pointers. Attached are the configuration files I used for my build and the hex file that works on the simple wood 1405. Extensions changed to .txt

[Configuration.txt](https://github.com/Printrbot/printrboardmodernmarlin/files/4093644/Configuration.txt)
[Configuration_adv.txt](https://github.com/Printrbot/printrboardmodernmarlin/files/4093647/Configuration_adv.txt)
[RevD-v.04-Simple Wood-XminYminZmin.txt](https://github.com/Printrbot/printrboardmodernmarlin/files/4093655/RevD-v.04-Simple.Wood-XminYminZmin.txt)

closed 2020-01-21 by Bointon