Printrbot/printrboardmodernmarlin #36: Heated Bed not working correctly after update

ZeroMavr1k, 2020-03-22:
After updating printrbot simple metal. My heated bed when at temperature consistently turns on and off rapidly I can hear the relay clicking on and off on and very quickly. I did not have this issue prior to updating heated bed worked fine.

Any thoughts?

Zakattacks, 2020-03-22:
The firmware is designed to use the integrated MOSFET in the Printrboard (See this article for more:[Reprap forums:Using relays for hotend and bed gives unexpected results](https://reprap.org/forum/read.php?415,767585))-- To use a relay, you have to change the config and re-compile as seen here: [MarlinFirmware Issue 267](https://github.com/MarlinFirmware/Marlin/issues/267)
Best solution is to upgrade to an external MOSFET if you're not able to use the built in one. A standard DC SSR (Solid State Relay) such as the ubiquitous SSR-25-DD should work fine.

ZeroMavr1k, 2020-03-23:
Ah I see. Thanks for the response. I am currently using a different power supply for my heated bed.
So I guess i have some figuring out to do.

Zakattacks, 2020-03-23:
The SSR would be a drop-in replacement for the relay in your relay. You might need a heatsink for it.