Printrbot/printrboardmodernmarlin #39: Printrbot Play HB w/ RepRapDiscount Smart Controller support

davidbrown-vsolvit, 2020-04-29:
I'm trying to compile my own hex for a Printrbot Play with the heated bed upgrade and a RepRapDiscount Smart Controller using PlatformIO. I'm really getting nowhere. Could someone compile a hex for me and upload it to the repository? I'd greatly appreciate it.
Cheers!

drphil3d, 2020-04-29:
The processor on the Printrboard only has 120k useable program memory, with all the features that give better print quality and ease of use you have to sacrifice some things. There isn't room for the display and SD card support.

davidbrown-vsolvit, 2020-04-29:
Understood. Thank you Phillip.

drphil3d, 2020-04-29:
What I recommend is using a raspberry pi with octoprint. The baby stepping feature is probably the most important of all.

closed 2020-04-29 by drphil3d