Printrbot/printrboardmodernmarlin #42: Issues compiling firmware

ritvikjay, 2020-07-07:
I am trying to compile the firmware for my Simple Metal HB no LCD with Bilinear Bed Leveling enabled but I am facing some issues. I don't think the issue is correlated with the bed leveling option I selected because I can't compile the unedited files by themselves. I have attached the log messages that I saw for reference. Thanks.

[ErrorLog.txt](https://github.com/Printrbot/printrboardmodernmarlin/files/4887272/ErrorLog.txt)

drphil3d, 2020-07-07:
you need to download the latest version of marlin (2.0.5.3 or later) they fixed this issue a few months ago.

ritvikjay, 2020-07-07:
When I open the folder I made in marlin-config it seems to auto download the Marlin firmware. Is there a way to ensure I have the latest version? Sorry for the confusion

drphil3d, 2020-07-07:
https://github.com/MarlinFirmware/Marlin/tree/2.0.5.3
This is the latest version, download the repo as a zip and just replace the files that are extracted with the ones that marlin-config downloads

ritvikjay, 2020-07-08:
I tried it again with pasting the extracted files into the Marlin folder for marlin-config but I'm still facing errors when compiling. Is there a specific version of the configuration files that I need to be using or is it a different issue? 

[ErrorLog1.pdf](https://github.com/Printrbot/printrboardmodernmarlin/files/4887672/ErrorLog1.pdf)

drphil3d, 2020-07-08:
don't click the compile button, use the drop down menu next to it and choose at90usb1286 dfu

ritvikjay, 2020-07-08:
Sorry, I accidentally forgot to do that for the file I sent last time. I tried compiling again with the latest release of the configuration files here with the latest version of Marlin with the correct board selected but faced errors when it got to stepper_dac.cpp

ritvikjay, 2020-07-09:
I think the above issue(stepper_dac.cpp) might have been not on the latest version of Marlin. When I tried it on the latest version I had an error saying select arduino mega or mega 2560 in the pins_RAMPS.h file even when compiling for the at90usb1285dfu board.

closed 2020-07-10 by ritvikjay